kaii-lb/overskride #3: Unit dbus-org.bluez.service not found

gmankab, 2023-10-17:
[I] ➜  ~ flatpak run io.github.kaii_lb.Overskride

(overskride:2): Gtk-WARNING **: 01:52:01.673: Unknown key gtk-modules in /home/gmanka/.var/app/io.github.kaii_lb.Overskride/config/gtk-4.0/settings.ini

(overskride:2): Gtk-WARNING **: 01:52:01.826: 'center' is not a valid child type of 'AdwHeaderBar'
thread 'main' panicked at src/window.rs:187:40:
cannot start presetup, something got REALLY fucked: Error { kind: Internal(DBus("org.freedesktop.systemd1.NoSuchUnit")), message: "Unit dbus-org.bluez.service not found." }
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

kaii-lb, 2023-10-18:
> also a am getting same error if i plug out the bluetooth dongle

yes ofcourse you would, the entire app depends on bluetooth existing and being enabled by systemd kr wtv service manager you use.
later on i plan to add a message for this special case, but for now you a bluetooth app wont work without bluetooth-

ill keep this open as a reminder

kaii-lb, 2023-10-29:
this is done, closing

closed 2023-10-29 by kaii-lb