kaii-lb/overskride #7: Device deselects sporadically

keis, 2023-10-18:
First, tool looks super slick! :)

I opened the application in the office where I have up towards 100 devices visible around me that also come and go quickly. List updates maybe once every other second or so.

Most show up as "Unknown" (I don't know what they are either) but I found my headset and selected it. However I was not able to interact with the interface before the list updated which also deselected the device. It doesn't happen on every update though so maybe it somehow got missed in one update for some reason.

kaii-lb, 2023-10-19:
ohhhhh yes i know what you are talking about, im trying to resolve the issue currently

the "unknown device" is usually like smart lights or doorbells or other stuff that uses bluetooth, but doesn't have a "name" assigned to it. i'll add a "more info" option to see more stuff about these, and sort them to the bottom

kaii-lb, 2023-10-20:
should have gotten fixed in the latest release, couldn't test it tho

kaii-lb, 2023-10-20:
alright i was able to test it now and figured out it don't work, pushed a new release that *does* infact work as expected
please test before i mark as closed

AshtakaOOf, 2023-10-22:
I installed [6599093662](https://github.com/kaii-lb/overskride/actions/runs/6599093662) but I have the same issue as keis while following the instructions in the readme.
Do keep in mind that I have only ever connected 3 Bluetooth devices on this Fedora install.
Gnome settings detects only my earbuds because I have removed my speaker and dualshock4 from my devices.

kaii-lb, 2023-10-22:
i cannot seem to replicate the issue at all, are you sure the device removed is _different_ from the one currently selected?

AshtakaOOf, 2023-10-22:
I get this on the last CI (0.4.6)
And the devices do deselect randomly after being clicked on.
[image]
I do plan on replacing my mt7921e with an ax210 because it crashes a lot.

kaii-lb, 2023-10-22:
alright, the chance of the device having exactly the same MAC Address as your earbuds is so extremely low, but i added another check for the name of the device just incase thats somehow happening with you, v0.4.7 is out, waiting for it to compile now (5 mins the CI build should be done)

kaii-lb, 2023-10-23:
i did notice that your device is getting sorted WAY at the bottom, that shouldn't be happening, you sure its the latest build? (this hasn't been the case since 0.4.4 i believe)

AshtakaOOf, 2023-10-24:
I've changed my wifi/bluetooth card from a mt7921e to an ax210.
Basically I live in apartment and it seems to be showing every single Bluetooth devices around me.
I don't know if it's the last version or my new chip, but it isn't deselecting my device anymore and it's shown at the top.
[image]

keis, 2023-10-24:
Thx, I tried the old version at home now and was not able to reproduce.

I'll try both versions next time I'm in a bluetooth noisy place

closed 2023-10-24 by kaii-lb

keis, 2023-10-26:
Yes, I confirm the issue seems to be gone now