kaii-lb/overskride #12: 'center' is not a valid child type of 'AdwHeaderBar'

roziscoding, 2023-10-21:
I just installed the app with flatpak as per the README instructions but, upon trying to run it with flatpak, I get the following:

<details>

<summary>
Full stack trace

```sh
❯ RUST_BACKTRACE=full flatpak run io.github.kaii_lb.Overskride

(overskride:2): Gtk-WARNING **: 14:00:54.459: 'center' is not a valid child type of 'AdwHeaderBar'

(overskride:2): Gtk-WARNING **: 14:00:54.461: 'center' is not a valid child type of 'AdwHeaderBar'
thread 'main' panicked at src/window.rs:1439:93:
cannot get user download dir
```

</summary>

```sh
stack backtrace:
   0:     0x56030db513bc - <unknown>
   1:     0x56030db77c2c - <unknown>
   2:     0x56030db4e8ce - <unknown>
   3:     0x56030db511a4 - <unknown>
   4:     0x56030db52853 - <unknown>
   5:     0x56030db5256c - <unknown>
   6:     0x56030db52dd9 - <unknown>
   7:     0x56030db52cd7 - <unknown>
   8:     0x56030db518e6 - <unknown>
   9:     0x56030db52a22 - <unknown>
  10:     0x56030d52c873 - <unknown>
  11:     0x56030d52c83a - <unknown>
  12:     0x56030d5564cc - <unknown>
  13:     0x56030d5f9fbe - <unknown>
  14:     0x56030d6da904 - <unknown>
  15:     0x56030d71da3b - <unknown>
  16:     0x7f895831795a - <unknown>
  17:     0x7f8958318ff6 - g_object_new_with_properties
  18:     0x56030db1b19a - <unknown>
  19:     0x56030db1aa7f - <unknown>
  20:     0x56030d72b908 - <unknown>
  21:     0x56030d6ef19a - <unknown>
  22:     0x56030d62281a - <unknown>
  23:     0x56030d631b6b - <unknown>
  24:     0x7f8958311652 - <unknown>
  25:     0x7f89583274af - <unknown>
  26:     0x7f895832d3c1 - g_signal_emit_valist
  27:     0x7f895832d483 - g_signal_emit
  28:     0x7f89585a1888 - <unknown>
  29:     0x56030d538b90 - <unknown>
  30:     0x56030d538193 - <unknown>
  31:     0x56030d631ee2 - <unknown>
  32:     0x7f89585a19eb - g_application_run
  33:     0x56030d5b0c02 - <unknown>
  34:     0x56030d5b0d19 - <unknown>
  35:     0x56030d5a5702 - <unknown>
  36:     0x56030d58528b - <unknown>
  37:     0x56030d54994e - <unknown>
  38:     0x56030d72a9d1 - <unknown>
  39:     0x56030db4aedb - <unknown>
  40:     0x56030d72a9aa - <unknown>
  41:     0x56030d5a586e - <unknown>
  42:     0x7f895755208a - <unknown>
  43:     0x7f895755214b - __libc_start_main
  44:     0x56030d52cfe5 - <unknown>
  45:                0x0 - <unknown>
```

</details>

I'm on EndeavourOS with i3-wm, but I do have a working Gnome setup too. I don't really which package versions to provide, but I'll gladly add any one you ask for :)

kaii-lb, 2023-10-21:
alright i pushed a new release, it should fix your issue

tho be mindful of the "received file location" setting, since if gtk wasn't able to find your download folder im not sure it would be able to find your home folder either, so you might end up having to manually select the folder

roziscoding, 2023-10-22:
I'll try it out in a bit

roziscoding, 2023-10-24:
It works! 😄 
Is there a way to set the app to use a dark theme, tho?

closed 2023-10-24 by kaii-lb