kaii-lb/overskride #21: Allow controlling multiple Bluetooth devices

acerspyro, 2024-02-05:
This is an extension of https://gitlab.gnome.org/GNOME/gnome-control-center/-/issues/2882

It's possible you're using the same backend as GNOME CC and would face the same limitations, but it would be useful to have if it is possible.

---

*Original issue*

I just discovered that the Wi-Fi section now lets you select which adapter you would like to interact with:

[image]

It's a very nice feature, and I'd like to see the same thing for Bluetooth adapters.

I have an AX210 chipset that does both Bluetooth and Wi-Fi. It works very well, but it has a bug with wireless Xbox controllers, so I have to instead use an ASUS USB-BT500 adapter for these devices.

It's currently impossible to interact with one adapter when the other is displayed in Settings unless I unplug the currently displayed adapter.

[image]

(I had the AX210 and the BT500 connected when I took the screenshot).

kaii-lb, 2024-02-05:
thats already implemented tho? check the Bluetooth Settings page on a recent build of overskride. it has that option built in (but reads info from each adapter so devices connected on one adapter wont show as connected on the second)

kaii-lb, 2024-02-05:
[image]

acerspyro, 2024-02-17:
Oh damn I'm blind, thank you!

I do think it would be more visible as tabs rather than under Bluetooth Settings though. We can close this issue and open a new one if you're open to that idea :)

closed 2024-02-17 by acerspyro