kaii-lb/overskride #23: Build artifact expired

alyssadev, 2024-03-24:
No downloads are available as indicated in README, as actions build artifacts have expired.

kaii-lb, 2024-03-24:
ohhh i did not notice that. i'll fix this in a sec. thank you for pointing it out.

kaii-lb, 2024-03-24:
check the latest release and please report if it works or not

alyssadev, 2024-03-24:
https://github.com/kaii-lb/overskride/actions/runs/8408171165 The build failed

kaii-lb, 2024-03-24:
not the build, but the releases page

alyssadev, 2024-03-25:
This build is amd64, the device I wanted to run it on is arm64. I'll see if I can fix the build process and get it working and PR a fix for main.yml that provides multi arch support. The lack of binary download has been resolved so I'll close this issue

closed 2024-03-25 by alyssadev