kaii-lb/overskride #25: Hyprland/NixOS: toggles seem to do nothing

TheRealGramdalf, 2024-05-08:
With my current NixOS configuration, Overskride seems unable to change any bluetooth properties - powering on/off the adapter has no effect. Other programs (`blueberry`, `bluetuith`) are able to power on the adapter, but opening overskride and toggling "connect" with any device does nothing.

Relevant nixos config:
```
  hardware = {
    bluetooth = {
      enable = true;
      powerOnBoot = false;
    };
  };
```

Starting overskride via the command line does not seem to give any errors (apart from some GTK ones)

Any suggestions as to what I could do from here?

kaii-lb, 2024-05-09:
you need to click on the row not the switch. gtk has a weird thing where clicking on the switch of the row wont do anything, but clicking the row itself will.

TheRealGramdalf, 2024-05-15:
I'll leave this open as this needs to be addressed, but the workaround does work for now. 

Out of curiosity, where are the relevant GTK docs/overskride source code that doesn't work? I find it very surprising that this wouldn't work, as I'm fairly certain it did on GNOME (which uses GTK4 for at least some circle apps)

kaii-lb, 2024-05-16:
maybe gnome does something to the switch itself, if you can find anything about that i'd be extremely grateful. 

here's *all* the relevant source code for overskride https://github.com/kaii-lb/overskride/blob/main/src/gtk/window.blp

Zetta1Reid0, 2024-06-23:
GNOME doesn't do anything to the switches. I think the problem is that the switches actually don't do anything. I don't know rust, but from what I have seen in the src, there is nowhere specified that when the switch is toggled some code should be executed, all I have seen is things flipping the switch.

sleklere, 2026-09-01:
Ran into this locally and went digging — it's not libadwaita being weird, and
you don't need to redesign the row.

`GtkSwitch` has its own click gesture and it claims the event sequence, so
clicking the switch never activates the row and `activated` never fires. The
switch still slides over, which is why it looks like the click worked.
`notify::active` fires wherever you click and carries the new value.

PR at #59.