kaii-lb/overskride #38: [FEATURE REQUEST]: Hide devices

5p4r74cu5, 2025-02-17:
Hey there, I would like to be able to hide devices, or a setting to hide unknown devices. I live in an apartment building and there are a squillion devices every time I open overskride.

5p4r74cu5, 2025-02-18:
oh my apologies, i forgot that the flatpak wasn't automatically updating to the latest version due to the manual installation. just updated to latest version and see the option your talking about.

how come overskride isnt on flathub?

kaii-lb, 2025-02-18:
> how come overskride isnt on flathub?
cuz i ran out of motivation for this project 😅 

> ive still got heaps of known devices that belong to other apartments
yeah that is an issue i suppose. i'll try to do something about it when i can

5p4r74cu5, 2025-02-19:
> cuz i ran out of motivation for this project 😅

Oh that's understandable; life happens. Well kudos for the work that you managed to do whilst motivated; its a solid program :-)

hotsno, 2025-06-01:
For me the issue was that the "Hide Unknown Devices" option was flaky. I'd enable it and nothing would happen, then after the restarting the app the setting was off again. After a couple retries the option stayed and now it works.